wuxx/Colorlight-FPGA-Projects #1: Design files for ext-board?

mcheah, 2020-12-28:
Hi,
Can you provide the full design files for the ext-board?  I'd like to modify to add a few peripherals without adding a pmod.  Thanks.

wuxx, 2020-12-30:
Hi, the origin design files is not open source, but willing share for personal use. leave your email and I will send you the origin design files.

closed 2020-12-30 by wuxx

mcheah, 2020-12-30:
Thanks man.
Email is 
Kulivontot@outlook.com

mcheah, 2021-01-09:
Hi @wuxx,
Can you share some details on the daplink microcontroller firmware on this board?  Is this just a stock daplink image or did you do anything custom to get it to work with the ECP5?  If you could share the source for the daplink that'd be outstanding.

wuxx, 2021-01-09:
Hi, mcheah, the on board DAPLink  base on the open source DAPLink project, click [here](https://github.com/ARMmbed/DAPLink).
or you can use my project [here](https://github.com/wuxx/nanoDAP/tree/master/software/v2.3a/DAPLink/projectfiles/uvision), I only add the jtag support (some IO config, not much modify)

mcheah, 2021-01-09:
Thanks for the links.
Do you know if the DAPlink implementation is supported by the official lattice tools (Diamond, etc)?  Or am I limited to using the IceStorm open source toolchain?

wuxx, 2021-01-09:
of course lattice IDE don't support download with DAPLink, the DAPLink is mainly designed for flash ARM Cortex-M based series MCU.
if you use under windows, can try this script.
https://github.com/wuxx/openocd-toolbox/tree/master/scripts/windows/colorlight-i5

hjcho1256, 2023-08-15:
Could I also receive the design files for the ext-board with hjcho1256@gmail.com? I'm working on making a board that will have the ext-board mounted on it. Thank you!

wuxx, 2023-08-16:
OK,  already sent.

MLtoPWELC, 2024-03-25:
> Hi, mcheah, the on board DAPLink base on the open source DAPLink project, click [here](https://github.com/ARMmbed/DAPLink). or you can use my project [...]

Can we change from STM32 internal flash to external flash for FPGA by simply modifying the JTAG port configuration without modifying the flash algorithm? In addition, the source code provides different flash algorithms and firmware for different series of STM32 chips. May I ask which series of code you modified from?I would greatly appreciate it if you could answer me.

jrfo-hwit, 2024-10-26:
Hi  wuxx, could you share the design files with me at jrfoliveira@gmail.com ? i am going to use it in a university fpga study group.