wuxx/Colorlight-FPGA-Projects #3: schematics for smaller board with dual ethernet

fanoush, 2021-04-25:
hello, you also sell this board https://www.aliexpress.com/item/1005002080383026.html for I5. I already ordered the bigger expansion board but this looks interesting too, however I did not find any description of the connectors. Are they all just outputs behind 5v level shifters like on other colorlight boards or are some connectors also bidirectional fpga pins without any level shifters?  Is there schematics for this board?

wuxx, 2021-04-26:
hello, there dual ethernet extboard has no schematic, it's a commercial product (not dev board), but the i5 document should enough, the etherenet pins are already reversed in README.md

fanoush, 2021-04-26:
thanks for reply, i5 document or ethernet pins does not help with my question about level shifters on those 4 big GPIO connectors, e.g. on 5A-75B these are all outputs behind 5V level shifters, here there are lot of level shifters on that board too but I am not sure if all 4 are like that since the ones on the right has no level shifter near it. but I guess it will be same as on other boards so unusable for inputs. otherwise it would be nice board (small size, ethernet and gpios on same board)

wuxx, 2021-04-26:
the GPIO on this ethernet board is also behind 5V level shifters, so it can only use to output, I think it's not very useful, only the dual ethernet is interest, you can use this bitstream https://github.com/wuxx/Colorlight-FPGA-Projects/blob/master/tools/pin_scan_bga381.bit to reverse the pins, it can output pin name with uart baudrate 115200, for example in A1 pin it will continuous output 'A1\r\n'

closed 2021-05-02 by wuxx