wuxx/Colorlight-FPGA-Projects #15: Which board to select?

zapta, 2024-02-12:
For the I9 board from AliExpress, which board do I need to choose in IceStudio?

[image]

[image]

wuxx, 2024-02-13:
Hello, none of these three are correct. In fact, the downloader is based on DAPLink,  For details, please check the README in this repo.

zapta, 2024-02-13:
Thanks for the answer, but I don't understand.

1. What board do I need to select in IceStudio for "I9 and Ext Board" ?

2. Can you give me a link of the documentation of the I9/Ext DAPLink and how to use it?

wuxx, 2024-02-20:
hello, please check here
https://github.com/wuxx/Colorlight-FPGA-Projects/blob/master/get-start.md

zapta, 2024-02-20:
Thanks. I think I understand it now. Your boards require an external FTDI
adapter to be compatible with apio or IceStudio. They explain it here
https://github.com/benitoss/ColorLight_FPGA_boards and say that using
FTDI speeds up the transfer significantly.