wuxx/Colorlight-FPGA-Projects #17: HDMI out of i9+ Ext board not routed to working TDMS balls on xc7a50t

exitrip, 2024-05-04:
I had to give up on openXC7 toolchain to build larger projects, but Vivado is providing clear errors:

D0+ cannot route to AB6:
[image]

D2+ cannot route to W7
[image]

Clk/D3+ cannot route to AB7 (which is actually the negative out of AA5):
[image]

I'm still looking for an actual datasheet that details the Diff Pair outputs of the xc7a50tfgg484, well enough to work from anything outside of Vivado.

If you want to provide the Design files (either Altium or Kicad) for the Ext board, I'm happy to fix the HDMI pinout myself.
Otherwise this issue is just a note that the HDMI connector does quite work on the Ext board.

exitrip, 2024-05-04:
This pinout looks like everything is broken out on the DIMM connector close to the current HDMI connector:
[image]

wuxx, 2024-05-05:
Hello, in fact HDMI can be implemented by ordinary GPIO, you can refer to here
https://github.com/wuxx/Colorlight-FPGA-Projects/tree/master/src/i5/hdmi_test_pattern

exitrip, 2024-05-05:
Truly, but using the on-chip hardware will enable higher resolutions and multiple video channels.

wuxx, 2024-05-06:
understand, the current design is for forward compatibility (colorlight i5 i9)

closed 2024-05-14 by wuxx