wuxx/Colorlight-FPGA-Projects #23: Please route tracks to DQM0-DQM3 pins on the SDRAM

jbilander, 2025-04-17:
Feature request:

Please for next revision of this board route tracks to the DQM0-DQM3 pins on the SDRAM, then this module can be useful as an accelerator in retro computers. For example implementing a 68k soft core in the FPGA and with a SDRAM controller where you can control and mask off bytes written to the SDRAM.

Regards, J

Disasm, 2025-04-17:
The boards with SDRAM are produced by Colorlight, with their own use case in mind. You may ask them of course, but I'm not sure if they are even aware of using their boards for purposes other than LED display control.

jbilander, 2025-04-17:
Ah, I see, I thought this was their github repository but ok I will try and contact them.

Thanks

closed 2025-04-21 by wuxx